ucb-bar/esp-llvm #15: Anyone pushing to get RISCV in mainline llvm/clang?

marcopeereboom, 2015-08-12:
I am trying to port the riscv stuff to the bitrig operating system.  We have llvm/clang in our base system and it would make the porting effort much easier if riscv becomes part of the mainline llvm/clang distribution.  By having this in base llvm/clang it prevents bitrot and divergent code bases.

Is someone working on doing this?

Is there any interest in doing this?

colinschmidt, 2015-08-12:
I am keeping the riscv-trunk branch up to date with upstream llvm/clang every few weeks.

There are a few other features that we need to have before we can start the upstreaming process (assembler comes to mind, but I think a few others). I will continue to work on these, others are welcome to help, and it does seem like talking with people there is interest in this.

marcopeereboom, 2015-08-12:
I asked because looking at the checkout it is version 3.3 and it misses the bitrig bits that are upstreamed.  Am I missing something?

colinschmidt, 2015-08-12:
This is the branch that is closest to upstream https://github.com/riscv/riscv-llvm/tree/riscv-trunk Its somewhere after 3.6 before 3.7. Let me know if that has the bitrig bits you need

marcopeereboom, 2015-08-12:
aha! I am on the main branch; that explains it.  Thanks!

I am still very interested in seeing this become mainlined :)

closed 2015-08-12 by marcopeereboom