ucb-bar/esp-llvm #18: Infinite loop when compile spec with clang

jakkdu, 2015-10-30:
I tried to compile SPEC2006 with https://github.com/ccelio/Speckle
When I compiled it, clang goes into infinite loop.
Below command is that I used to compile perl in SPEC2006.
Infinite loop is caused in RISCVBranchSelector.cpp

```sh
clang -S -target riscv -mriscv=RV64IAMFD -I/lib/gcc/riscv64-unknown-linux-gnu/5.2.0//include -I/lib/gcc/riscv64-unknown-linux-gnu/5.2.0//include-fixed -I/lib/gcc/riscv64-unknown-linux-gnu/5.2.0//../../../../riscv64-unknown-linux-gnu/include -I/sysroot/usr/include -static -std=gnu89 -Wl,-Ttext-segment,0x10000 -DSPEC_CPU -DNDEBUG -DPERL_CORE -fno-strict-aliasing -DSPEC_CPU_LP64 -DSPEC_CPU_LINUX_X64 -o regexec.S regexec.c -O0 
````

colinschmidt, 2015-10-30:
Which branch of the repo did you use?

jakkdu, 2015-10-31:
I used RISCV.

colinschmidt, 2015-10-31:
I think that bug was fixed on riscv-trunk and I was pretty sure I back ported it to RISCV. I'll double check that but it might be better for you to just use riscv-trunk if you don't have a reason to use an older version of llvm.

jakkdu, 2015-11-02:
I tried to use riscv-trunk. But when I compiled sample(hello.c) it gives me such error.
```
./install/bin/clang -target riscv -mriscv=RV64IAMFD -S hello.c -o hello.S
error: unable to create target: 'No available targets are compatible with this triple, see -version for the available targets.'
```

I used riscv-clang, if I put it random string as target it gives me different error
```
clang: warning: argument unused during compilation: '-mriscv=RV64IAMFD'
error: unknown target triple 'asdf', please use -triple or -arch
```

I disabled bindings but I think it is not related with this issue.
```
../configure --disable-bindings
make
make install
```
Did I do something wrong?

colinschmidt, 2015-11-02:
I think it should be 
```
clang -target riscv64 ...
```

jakkdu, 2015-11-02:
Sorry. Not working in my machine.

```
./install/bin/clang -target riscv64 -mriscv=RV64IAMFD -S hello.c -o hello.S
error: unable to create target: 'No available targets are compatible with this triple, see -version for the available targets.'
1 error generated.
```

colinschmidt, 2015-11-02:
What does 
```
llvm-config --targets-built
```

return? I usually explicitly enable riscv in the configure line like
```
../configure --enable-targets=riscv
```

neuschaefer, 2015-11-02:
Does it work with `-mriscv=RV64IMAFD` instead of `-mriscv=RV64IAMFD`?

jakkdu, 2015-11-02:
Thanks. Hello World example can be works when I configure with --enable-targets=riscv.
However, spec cannot be compiled. 
When I am using riscv-trunk, I got another assertion error and clang seems to go to infinite loop when compiling regexec.c.

This is the code that gets assertion error.
```
Instruction uses an allocated register
UNREACHABLE executed at /home/insu/projects/hdfi/src/riscv-llvm/lib/CodeGen/RegAllocFast.cpp:361!
#0 0x3d9586e llvm::sys::PrintStackTrace(llvm::raw_ostream&) /home/insu/projects/hdfi/src/riscv-llvm/lib/Support/Unix/Signals.inc:404:15
#1 0x3d968a9 PrintStackTraceSignalHandler(void*) /home/insu/projects/hdfi/src/riscv-llvm/lib/Support/Unix/Signals.inc:463:1
#2 0x3d94fc3 llvm::sys::RunSignalHandlers() /home/insu/projects/hdfi/src/riscv-llvm/lib/Support/Signals.cpp:35:5
#3 0x3d99909 SignalHandler(int) /home/insu/projects/hdfi/src/riscv-llvm/lib/Support/Unix/Signals.inc:211:1
```

Is there someone who successfully compile SPEC2006 with clang?