ucb-bar/esp-llvm #20: Failure trying to build GNU coreutils with Clang

jwiegley, 2015-11-13:
I'm trying to build GNU coreutils, the `wc` utility in particular, and I'm seeing the exception below.  The code raising it is in `RegAllocFast.cpp:361`:

``` c++
  switch (PhysRegState[PhysReg]) {
  case regDisabled:
    break;
  case regReserved:
    PhysRegState[PhysReg] = regFree;
    // Fall through
  case regFree:
    MO.setIsKill();
    return;
  default:
    // The physreg was allocated to a virtual register. That means the value we
    // wanted has been clobbered.
    llvm_unreachable("Instruction uses an allocated register");
  }
```

The trace is:

```
johnw@ubuntu:~/coreutils-8.24$ make src/wc
  CC       lib/open-safer.o
Instruction uses an allocated register
UNREACHABLE executed at /home/johnw/riscv-llvm/lib/CodeGen/RegAllocFast.cpp:361!
#0 0x3d4f39e llvm::sys::PrintStackTrace(llvm::raw_ostream&) /home/johnw/riscv-llvm/lib/Support/Unix/Signals.inc:405:0
#1 0x3d4f6b3 PrintStackTraceSignalHandler(void*) /home/johnw/riscv-llvm/lib/Support/Unix/Signals.inc:463:0
#2 0x3d4dde6 llvm::sys::RunSignalHandlers() /home/johnw/riscv-llvm/lib/Support/Signals.cpp:34:0
#3 0x3d4e2be SignalHandler(int) /home/johnw/riscv-llvm/lib/Support/Unix/Signals.inc:211:0
#4 0x2aedbaaa3340 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x10340)
#5 0x2aedbb70bcc9 gsignal (/lib/x86_64-linux-gnu/libc.so.6+0x36cc9)
#6 0x2aedbb70f0d8 abort (/lib/x86_64-linux-gnu/libc.so.6+0x3a0d8)
#7 0x3d334e5 bindingsErrorHandler(void*, std::string const&, bool) /home/johnw/riscv-llvm/lib/Support/ErrorHandling.cpp:126:0
#8 0x332c36a (anonymous namespace)::RAFast::usePhysReg(llvm::MachineOperand&) /home/johnw/riscv-llvm/lib/CodeGen/RegAllocFast.cpp:351:0
#9 0x332f428 (anonymous namespace)::RAFast::AllocateBasicBlock() /home/johnw/riscv-llvm/lib/CodeGen/RegAllocFast.cpp:945:0
#10 0x333001e (anonymous namespace)::RAFast::runOnMachineFunction(llvm::MachineFunction&) /home/johnw/riscv-llvm/lib/CodeGen/RegAllocFast.cpp:1092:0
#11 0x32da0c7 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) /home/johnw/riscv-llvm/lib/CodeGen/MachineFunctionPass.cpp:41:0
#12 0x3c34baa llvm::FPPassManager::runOnFunction(llvm::Function&) /home/johnw/riscv-llvm/lib/IR/LegacyPassManager.cpp:1520:0
#13 0x3c34d27 llvm::FPPassManager::runOnModule(llvm::Module&) /home/johnw/riscv-llvm/lib/IR/LegacyPassManager.cpp:1540:0
#14 0x3c3507d (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) /home/johnw/riscv-llvm/lib/IR/LegacyPassManager.cpp:1596:0
#15 0x3c3575b llvm::legacy::PassManagerImpl::run(llvm::Module&) /home/johnw/riscv-llvm/lib/IR/LegacyPassManager.cpp:1698:0
#16 0x3c35995 llvm::legacy::PassManager::run(llvm::Module&) /home/johnw/riscv-llvm/lib/IR/LegacyPassManager.cpp:1730:0
#17 0x15913d2 (anonymous namespace)::EmitAssemblyHelper::EmitAssembly(clang::BackendAction, llvm::raw_pwrite_stream*) /home/johnw/riscv-llvm/tools/clang/lib/CodeGen/BackendUtil.cpp:657:0
#18 0x159149c clang::EmitBackendOutput(clang::DiagnosticsEngine&, clang::CodeGenOptions const&, clang::TargetOptions const&, clang::LangOptions const&, llvm::StringRef, llvm::Module*, clang::BackendAction, llvm::raw_pwrite_stream*) /home/johnw/riscv-llvm/tools/clang/lib/CodeGen/BackendUtil.cpp:673:0
#19 0x156f5e2 clang::BackendConsumer::HandleTranslationUnit(clang::ASTContext&) /home/johnw/riscv-llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:186:0
#20 0x1892086 clang::ParseAST(clang::Sema&, bool, bool) /home/johnw/riscv-llvm/tools/clang/lib/Parse/ParseAST.cpp:170:0
#21 0x124b3ee clang::ASTFrontendAction::ExecuteAction() /home/johnw/riscv-llvm/tools/clang/lib/Frontend/FrontendAction.cpp:540:0
#22 0x1571dc0 clang::CodeGenAction::ExecuteAction() /home/johnw/riscv-llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:788:0
#23 0x124aeb1 clang::FrontendAction::Execute() /home/johnw/riscv-llvm/tools/clang/lib/Frontend/FrontendAction.cpp:443:0
#24 0x12094c6 clang::CompilerInstance::ExecuteAction(clang::FrontendAction&) /home/johnw/riscv-llvm/tools/clang/lib/Frontend/CompilerInstance.cpp:822:0
#25 0x11c962b clang::ExecuteCompilerInvocation(clang::CompilerInstance*) /home/johnw/riscv-llvm/tools/clang/lib/FrontendTool/ExecuteCompilerInvocation.cpp:222:0
#26 0x11b0bba cc1_main(llvm::ArrayRef<char const*>, char const*, void*) /home/johnw/riscv-llvm/tools/clang/tools/driver/cc1_main.cpp:116:0
#27 0x11c2b74 ExecuteCC1Tool(llvm::ArrayRef<char const*>, llvm::StringRef) /home/johnw/riscv-llvm/tools/clang/tools/driver/driver.cpp:380:0
#28 0x11c32cc main /home/johnw/riscv-llvm/tools/clang/tools/driver/driver.cpp:443:0
#29 0x2aedbb6f6ec5 __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21ec5)
#30 0x11aee59 _start (/opt/riscv/bin/clang+0x11aee59)
Stack dump:
0.      Program arguments: /opt/riscv/bin/clang -cc1 -triple riscv64-unknown--elf -S -disable-free -main-file-name open-safer.c -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -no-integrated-as -mconstructor-aliases -target-feature +rv64 -target-feature +a -target-feature +m -target-feature +f -target-feature +d -target-linker-version 2.24 -dwarf-column-info -coverage-file /home/johnw/coreutils-8.24/open-safer.c -resource-dir /opt/riscv/bin/../lib/clang/3.8.0 -I . -I ./lib -I lib -I ./lib -I src -I ./src -I /opt/riscv/riscv64-unknown-elf/include -fno-dwarf-directory-asm -fdebug-compilation-dir /home/johnw/coreutils-8.24 -ferror-limit 19 -fmessage-length 100 -mstackrealign -fobjc-runtime=gcc -fdiagnostics-show-option -o /tmp/open-safer-eea432.s -x c lib/open-safer.c
1.      <eof> parser at end of file
2.      Code generation
3.      Running pass 'Function Pass Manager' on module 'lib/open-safer.c'.
4.      Running pass 'Fast Register Allocator' on function '@open_safer'
clang: error: unable to execute command: Aborted (core dumped)
clang: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.8.0
Target: riscv64-unknown--elf
Thread model: posix
InstalledDir: /opt/riscv/bin
clang: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang: note: diagnostic msg: /tmp/open-safer-e9100b.c
clang: note: diagnostic msg: /tmp/open-safer-e9100b.sh
clang: note: diagnostic msg:

********************
make: *** [lib/open-safer.o] Error 254
```

jwiegley, 2015-11-13:
I configured the build using this command:

``` bash
./configure CC=/opt/riscv/bin/clang CFLAGS="-mriscv=RV64IAMFD -I/opt/riscv/riscv64-unknown-elf/include -target riscv64-unknown-elf" --host=riscv64-unknown-elf
```

I have the following built and installed on the Ubuntu 14.04 machine:

- riscv-gnu-toolchain
- riscv-isa-sim
- riscv-llvm
- riscv-pk
- riscv-poky
- riscv-qemu
- riscv-tests

jwiegley, 2015-11-17:
Any thoughts on what to look at to debug this failure?

neuschaefer, 2015-11-17:
[Delta](http://delta.tigris.org/) can probably help in getting a compact test case.

jwiegley, 2015-11-17:
For the time being, configuring with `-O2` changes the allocation sufficiently that the file in question  compiles.

jwiegley, 2015-11-18:
I tried reproducing this today without `-O2`, to narrow it down, but it's no longer happening.

closed 2015-11-18 by jwiegley