ucb-bar/esp-llvm #25: Out-of-bounds read using llvm-mc -show-inst-operands

neuschaefer, 2015-11-24:
```
$ echo j 0x10 | build/bin/llvm-mc -show-inst-operands -triple riscv
=================================================================
==783==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60800000ab00 at pc 0x00000050eaa5 bp 0x7ffdb86de190 sp 0x7ffdb86de188
READ of size 8 at 0x60800000ab00 thread T0
    #0 0x50eaa4 in (anonymous namespace)::RISCVAsmParser::parseRegister((anonymous namespace)::RISCVAsmParser::Register&) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build/../lib/Target/RISCV/AsmParser/RISCVAsmParser.cpp:545:7
    #1 0x50e29e in (anonymous namespace)::RISCVAsmParser::parseRegister((anonymous namespace)::RISCVAsmParser::Register&, char, unsigned int const*, bool) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build/../lib/Target/RISCV/AsmParser/RISCVAsmParser.cpp:603:7
    #2 0x50e29e in (anonymous namespace)::RISCVAsmParser::parseRegister(llvm::SmallVectorImpl<std::unique_ptr<llvm::MCParsedAsmOperand, std::default_delete<llvm::MCParsedAsmOperand> > >&, char, unsigned int const*, (anonymous namespace)::RISCVOperand::RegisterKind, bool) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build/../lib/Target/RISCV/AsmParser/RISCVAsmParser.cpp:627
    #3 0x5ebb1d in (anonymous namespace)::AsmParser::parseStatement((anonymous namespace)::ParseStatementInfo&, llvm::MCAsmParserSemaCallback*) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/../lib/MC/MCParser/AsmParser.cpp:1635:7
    #4 0x5d8142 in (anonymous namespace)::AsmParser::Run(bool, bool) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/../lib/MC/MCParser/AsmParser.cpp:654:10
    #5 0x4eeeba in AssembleInput(char const*, llvm::Target const*, llvm::SourceMgr&, llvm::MCContext&, llvm::MCStreamer&, llvm::MCAsmInfo&, llvm::MCSubtargetInfo&, llvm::MCInstrInfo&, llvm::MCTargetOptions&) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/../tools/llvm-mc/llvm-mc.cpp:363:13
    #6 0x4eeeba in main /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/../tools/llvm-mc/llvm-mc.cpp:527
    #7 0x7f74a0025b44 in __libc_start_main /tmp/buildd/glibc-2.19/csu/libc-start.c:287
    #8 0x4e8689 in _start (/home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/bin/llvm-mc+0x4e8689)

0x60800000ab00 is located 8 bytes to the right of 88-byte region [0x60800000aaa0,0x60800000aaf8)
allocated by thread T0 here:
    #0 0x4637bb in operator new(unsigned long) (/home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/bin/llvm-mc+0x4637bb)
    #1 0x4fde50 in _ZN4llvm11make_uniqueIN12_GLOBAL__N_112RISCVOperandEJNS2_11OperandKindERNS_5SMLocES5_EEENSt9enable_ifIXntsr3std8is_arrayIT_EE5valueESt10unique_ptrIS7_St14default_deleteIS7_EEE4typeEDpOT0_ /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build/../include/llvm/ADT/STLExtras.h:390:3
    #2 0x4fde50 in (anonymous namespace)::RISCVOperand::createToken(llvm::StringRef, llvm::SMLoc) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build/../lib/Target/RISCV/AsmParser/RISCVAsmParser.cpp:108
    #3 0x4fde50 in (anonymous namespace)::RISCVAsmParser::ParseInstruction(llvm::ParseInstructionInfo&, llvm::StringRef, llvm::SMLoc, llvm::SmallVectorImpl<std::unique_ptr<llvm::MCParsedAsmOperand, std::default_delete<llvm::MCParsedAsmOperand> > >&) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build/../lib/Target/RISCV/AsmParser/RISCVAsmParser.cpp:722
    #4 0x5eb495 in (anonymous namespace)::AsmParser::parseStatement((anonymous namespace)::ParseStatementInfo&, llvm::MCAsmParserSemaCallback*) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/../lib/MC/MCParser/AsmParser.cpp:1623:19
    #5 0x5d8142 in (anonymous namespace)::AsmParser::Run(bool, bool) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/../lib/MC/MCParser/AsmParser.cpp:654:10
    #6 0x4eeeba in AssembleInput(char const*, llvm::Target const*, llvm::SourceMgr&, llvm::MCContext&, llvm::MCStreamer&, llvm::MCAsmInfo&, llvm::MCSubtargetInfo&, llvm::MCInstrInfo&, llvm::MCTargetOptions&) /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/../tools/llvm-mc/llvm-mc.cpp:363:13
    #7 0x4eeeba in main /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build1/../tools/llvm-mc/llvm-mc.cpp:527
    #8 0x7f74a0025b44 in __libc_start_main /tmp/buildd/glibc-2.19/csu/libc-start.c:287

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/jn/dev/riscv/rocket-chip/riscv-tools/riscv-llvm/build/../lib/Target/RISCV/AsmParser/RISCVAsmParser.cpp:545 (anonymous namespace)::RISCVAsmParser::parseRegister((anonymous namespace)::RISCVAsmParser::Register&)
Shadow bytes around the buggy address:
  0x0c107fff9510: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c107fff9520: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c107fff9530: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c107fff9540: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff9550: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
=>0x0c107fff9560:[fa]fa fa fa 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff9570: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff9580: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff9590: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff95a0: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c107fff95b0: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Heap right redzone:      fb
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack partial redzone:   f4
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  ASan internal:           fe
==783==ABORTING
0  llvm-mc         0x000000000048b335 backtrace + 149
1  llvm-mc         0x00000000007012b2 llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 642
2  llvm-mc         0x00000000006ffc94 llvm::sys::RunSignalHandlers() + 164
3  llvm-mc         0x0000000000705e8f
4  libpthread.so.0 0x00007f74a0e698d0
5  libc.so.6       0x00007f74a0039107 gsignal + 55
6  libc.so.6       0x00007f74a003a4e8 abort + 328
7  llvm-mc         0x00000000004e1756
8  llvm-mc         0x00000000004d2447
9  llvm-mc         0x00000000004d8e5f __sanitizer::Die() + 15
10 llvm-mc         0x00000000004d0adb
11 llvm-mc         0x00000000004d0621 __asan_report_error + 2897
12 llvm-mc         0x00000000004d12f7 __asan_report_load8 + 39
13 llvm-mc         0x000000000050eaa5
14 llvm-mc         0x000000000050e29f
15 llvm-mc         0x00000000005ebb1e
16 llvm-mc         0x00000000005d8143
17 llvm-mc         0x00000000004eeebb main + 23339
18 libc.so.6       0x00007f74a0025b45 __libc_start_main + 245
19 llvm-mc         0x00000000004e868a
Stack dump:
0.	Program arguments: build/bin/llvm-mc -show-inst-operands -triple riscv 
```

(To build with ASan, I set the cmake variable `LLVM_USE_SANITIZER` to `Address`.)

neuschaefer, 2015-11-25:
Interestingly, the same bug is present in our version of the SystemZ backend.

colinschmidt, 2015-11-25:
Ah ok. I can't look into this quite yet but that helps. Thanks for the
report!

neuschaefer, 2015-11-26:
The (SystemZ version of the) bug is still in upstream LLVM. I reported it there: https://llvm.org/bugs/show_bug.cgi?id=25647