ucb-bar/esp-llvm #27: test/CodeGen/RISCV/jal.ll fails in a debug build

neuschaefer, 2015-12-11:
```
$ build/bin/llc < test/CodeGen/RISCV/jal.ll 
        .text
        .file   "<stdin>"

# Machine code for function f1: Post SSA

BB#0: derived from LLVM BB %entry
        %vreg1<def> = COPY %ra; GR32Bit:%vreg1
        %a0<def> = COPY %vreg1; GR32Bit:%vreg1
        RET %a0<imp-def,dead>, %a1<imp-def,dead>, %a0<imp-use,kill>

# End machine code for function f1.

*** Bad machine code: Using an undefined physical register ***
- function:    f1
- basic block: BB#0 entry (0x2b24c68)
- instruction: %vreg1<def> = COPY- operand 1:   %ra
LLVM ERROR: Found 1 machine code errors.
```

Since the machine code snippet uses `ra`, the error seems to be related to the `llvm.returnaddress` intrinsic, which is called in `jal.ll`.

colinschmidt, 2015-12-11:
I'm a little busy today so I don't know if I'll get to this today, but I wanted to let you know that I have some fixes in flight in similar to #26 including a fix to this issue. The pull request still has additions/fixes I haven't made yet so after I'll probably pull in some of your commits after I push my updates.

Thanks for your help!

colinschmidt, 2015-12-19:
This was fixed and the test re-enabled. Thanks for the report.

closed 2015-12-19 by colinschmidt