ucb-bar/esp-llvm #30: Upstream RISC-V changes to allow for a Rust-lang port

sunnystormy, 2016-06-17:
I just opened a [ticket](https://github.com/rust-lang/rust/issues/34333) on the Rust-lang repository regarding a RISC-V port and was informed that we need to upstream the changes in this repository in order to be able to compile Rust without any issues.

Please let me know if and when this can be done. Thank you!

boomshroom, 2016-06-23:
I've been working on a local copy of rustc-nightly to try and get rust code to work. Even after adding target files for riscv64 (I haven't attempted a 32 bit target) and merging the differences between rust-llvm and riscv-llvm, the process to get it to build was very convoluted.

I have just now been able to compile an empty rust library for riscv without stdcore and I still got error messages from llvm about:

`'riscv64' is not a recognized processor for this target (ignoring processor)`

though that is probably just an incorrect config file.

The big problem that I have run into when trying to compile something a little bigger (define 2 variables and set one to the other), LLVM complained:

`ExpandIntegerResult #0: t12: i64 = FrameIndex<1>

Do not know how to expand the result of this operator!
UNREACHABLE executed at /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/SelectionDAG/LegalizeIntegerTypes.cpp:1300!
#0 0x000000000195f7de llvm::sys::PrintStackTrace(llvm::raw_ostream&) /home/angelo-arch/src/riscv/riscv-llvm/lib/Support/Unix/Signals.inc:321:15
#1 0x000000000195fc89 PrintStackTraceSignalHandler(void*) /home/angelo-arch/src/riscv/riscv-llvm/lib/Support/Unix/Signals.inc:380:1
#2 0x000000000195de23 llvm::sys::RunSignalHandlers() /home/angelo-arch/src/riscv/riscv-llvm/lib/Support/Signals.cpp:45:5
#3 0x000000000195ff14 SignalHandler(int) /home/angelo-arch/src/riscv/riscv-llvm/lib/Support/Unix/Signals.inc:210:1
#4 0x00007f2f646f2f00 __restore_rt (/usr/lib/libpthread.so.0+0x10f00)
#5 0x00007f2f63426295 __GI_raise (/usr/lib/libc.so.6+0x33295)
#6 0x00007f2f634276da __GI_abort (/usr/lib/libc.so.6+0x346da)
#7 0x000000000193c810 LLVMInstallFatalErrorHandler /home/angelo-arch/src/riscv/riscv-llvm/lib/Support/ErrorHandling.cpp:132:0
#8 0x0000000000d2c151 llvm::DAGTypeLegalizer::ExpandIntegerResult(llvm::SDNode*, unsigned int) /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/SelectionDAG/LegalizeIntegerTypes.cpp:1300:5
#9 0x0000000000cc6b9a llvm::DAGTypeLegalizer::run() /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/SelectionDAG/LegalizeTypes.cpp:238:17
#10 0x0000000000ccce0e llvm::SelectionDAG::LegalizeTypes() /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/SelectionDAG/LegalizeTypes.cpp:1183:10
#11 0x0000000000c72a16 llvm::SelectionDAGISel::CodeGenAndEmitDAG() /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:758:15
#12 0x0000000000c7242f llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator<llvm::Instruction const>, llvm::ilist_iterator<llvm::Instruction const>, bool&) /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:669:1
#13 0x0000000000c71840 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:1362:5
#14 0x0000000000c6f3fe llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:507:36
#15 0x00000000007afc87 (anonymous namespace)::RISCVDAGToDAGISel::runOnMachineFunction(llvm::MachineFunction&) /home/angelo-arch/src/riscv/riscv-llvm/lib/Target/RISCV/RISCVISelDAGToDAG.cpp:228:8
#16 0x0000000000f3e4ae llvm::MachineFunctionPass::runOnFunction(llvm::Function&) /home/angelo-arch/src/riscv/riscv-llvm/lib/CodeGen/MachineFunctionPass.cpp:43:3
#17 0x00000000018645af llvm::FPPassManager::runOnFunction(llvm::Function&) /home/angelo-arch/src/riscv/riscv-llvm/lib/IR/LegacyPassManager.cpp:1537:23
#18 0x00000000018648c5 llvm::FPPassManager::runOnModule(llvm::Module&) /home/angelo-arch/src/riscv/riscv-llvm/lib/IR/LegacyPassManager.cpp:1558:16
#19 0x000000000186505a (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) /home/angelo-arch/src/riscv/riscv-llvm/lib/IR/LegacyPassManager.cpp:1614:23
#20 0x0000000001864b86 llvm::legacy::PassManagerImpl::run(llvm::Module&) /home/angelo-arch/src/riscv/riscv-llvm/lib/IR/LegacyPassManager.cpp:1717:16
#21 0x0000000001865681 llvm::legacy::PassManager::run(llvm::Module&) /home/angelo-arch/src/riscv/riscv-llvm/lib/IR/LegacyPassManager.cpp:1748:3
#22 0x00000000006c7b00 compileModule(char**, llvm::LLVMContext&) /home/angelo-arch/src/riscv/riscv-llvm/tools/llc/llc.cpp:405:5
#23 0x00000000006c6391 main /home/angelo-arch/src/riscv/riscv-llvm/tools/llc/llc.cpp:211:13
#24 0x00007f2f63413741 __libc_start_main (/usr/lib/libc.so.6+0x20741)
#25 0x00000000006c3999 _start (/home/angelo-arch/opt/riscv/bin/llc+0x6c3999)
Stack dump:
0.      Program arguments: llc lib.ll 
1.      Running pass 'Function Pass Manager' on module 'lib.ll'.
2.      Running pass 'RISCV DAG->DAG Pattern Instruction Selection' on function '@rust_main'
`

That was after compiling to llvm-ir and running llc, compiling rust to assembly directly emitted a much smaller error message for the same problem.

boomshroom, 2016-06-23:
Quick update: the compiler segfaulted.

The target that failed is

    x86_64-unknown-linux-gnu/stage2/lib/rustlib/riscv64-unknown-linux-gnu/lib/stamp.core

GDB says it was on line 46 of the llvm/lib/MC/MCELFStreamer.cpp. The odd thing about this is that it's using the llvm I pointed to with --llvm-root and is targeting risc-v.

The backtrace of the core dump is [here](http://pastebin.com/XBpVKSEU)

esclear, 2017-05-09:
Any update on this effort?

sorear, 2017-05-10:
This particular RISC-V port is unlikely to be upstreamed because @asb has [started upstreaming](https://reviews.llvm.org/differential/?authors=asb) a [different patch series](https://github.com/lowRISC/riscv-llvm).

daylightbrightledlight, 2017-08-24:
Needs to be updated with [lowRISC/riscv-llvm](https://github.com/lowRISC/riscv-llvm)?