ucb-bar/esp-llvm #36: Build fails immediately

vanjoe, 2016-08-30:
Hi, I tried to follow the instructions in the readme, and I got the frollowing error:

```
joel:/nfs/scratch/joel 
 $ git clone -b RISCV https://github.com/riscv/riscv-llvm.git
Cloning into 'riscv-llvm'...
remote: Counting objects: 1057149, done.
remote: Total 1057149 (delta 0), reused 0 (delta 0), pack-reused 1057149
Receiving objects: 100% (1057149/1057149), 206.89 MiB | 1.41 MiB/s, done.
Resolving deltas: 100% (866268/866268), done.
Checking connectivity... done.
Checking out files: 100% (11855/11855), done.
joel:/nfs/scratch/joel 
 $ cd riscv-llvm/
joel:/nfs/scratch/joel/riscv-llvm (RISCV)
 $ git submodule update --init --recursive 
Submodule 'riscv-clang' (https://github.com/riscv/riscv-clang.git) registered for path 'riscv-clang'
Cloning into 'riscv-clang'...
remote: Counting objects: 543424, done.
remote: Total 543424 (delta 0), reused 0 (delta 0), pack-reused 543424
Receiving objects: 100% (543424/543424), 102.46 MiB | 1.52 MiB/s, done.
Resolving deltas: 100% (456062/456062), done.
Checking connectivity... done.
Submodule path 'riscv-clang': checked out '5ca597d912b945857b457674db7ca80370eb9a7c'
joel:/nfs/scratch/joel/riscv-llvm (RISCV)
 $ mkdir build
joel:/nfs/scratch/joel/riscv-llvm (RISCV)
 $ cd build
joel:/nfs/scratch/joel/riscv-llvm/build (RISCV)
 $ cmake -DCMAKE_INSTALL_PREFIX=/nfs/scratch/joel/riscv-llvm/install -DLLVM_TARGETS_TO_BUILD="RISCV" ../
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:143 (message):
  The target `RISCV' does not exist.

      It should be one of

  
  AArch64;ARM;CppBackend;Hexagon;Mips;MBlaze;MSP430;NVPTX;PowerPC;Sparc;SystemZ;X86;XCore

-- Configuring incomplete, errors occurred!
See also "/nfs/scratch/joel/riscv-llvm/build/CMakeFiles/CMakeOutput.log".

```

comododragon, 2016-10-05:
I had the same problem. It appears that the commands in the README are wrong. The git clone command points to the RISCV branch, whereas the README is for the riscv-trunk branch. Use this instead:

git clone -b riscv-trunk https://github.com/riscv/riscv-llvm.git

colinschmidt, 2016-10-05:
Yes the only supported branch is currently riscv-trunk.

Git-Hub-2016, 2016-11-05:
riscv-trunk does not work either! There is something wrong with the content and/or location of CMakeLists.txt

Git-Hub-2016, 2016-11-05:
Here is the result:
CMake Error at CMakeLists.txt:116 (include):
  include could not find load file:

    VersionFromVCS

CMake Error at /usr/share/cmake-3.5/Modules/CPack.cmake:423 (message):
  CPack license resource file:
  "/home/llvm/FPGA/Original/riscv-llvm/install/LICENSE.TXT" could
  not be found.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CPack.cmake:428 (cpack_check_file_exists)
  CMakeLists.txt:156 (include)

CMake Error at CMakeLists.txt:282 (include):
  include could not find load file:

    AddLLVMDefinitions

CMake Error at CMakeLists.txt:466 (include):
  include could not find load file:

    config-ix

CMake Error at CMakeLists.txt:468 (string):
  string sub-command REPLACE requires at least four arguments.

CMake Error at CMakeLists.txt:478 (include):
  include could not find load file:

    HandleLLVMOptions

CMake Error at CMakeLists.txt:544 (message):
  Unexpected failure executing llvm-build: /usr/bin/python2.7: can't open
  file
  '/home/llvm/FPGA/Original/riscv-llvm/install/utils/llvm-build/llvm-build':
  [Errno 2] No such file or directory
-- Constructing LLVMBuild project information
-- Configuring incomplete, errors occurred!
See also "/home/llvm/FPGA/Original/riscv-llvm/CMakeFiles/CMakeOutput.log".

colinschmidt, 2016-11-05:
It looks like you don't have python installed

"Unexpected failure executing llvm-build: /usr/bin/python2.7: can't open
file"

nitish2112, 2016-11-08:
I cloned the riscv-trunk branch, but the CMake is still giving errors:

Determining if the include file malloc/malloc.h exists failed with the following output:
Change Dir: /home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_1ae93/fast"
/usr/bin/make -f CMakeFiles/cmTC_1ae93.dir/build.make CMakeFiles/cmTC_1ae93.dir/build
make[1]: Entering directory '/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1ae93.dir/CheckIncludeFile.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_1ae93.dir/CheckIncludeFile.c.o   -c /home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:27: fatal error: malloc/malloc.h: No such file or directory
compilation terminated.
CMakeFiles/cmTC_1ae93.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_1ae93.dir/CheckIncludeFile.c.o' failed
make[1]: *** [CMakeFiles/cmTC_1ae93.dir/CheckIncludeFile.c.o] Error 1
make[1]: Leaving directory '/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_1ae93/fast' failed
make: *** [cmTC_1ae93/fast] Error 2

Determining if the include file ndir.h exists failed with the following output:
Change Dir: /home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_cc8f8/fast"
/usr/bin/make -f CMakeFiles/cmTC_cc8f8.dir/build.make CMakeFiles/cmTC_cc8f8.dir/build
make[1]: Entering directory '/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_cc8f8.dir/CheckIncludeFile.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_cc8f8.dir/CheckIncludeFile.c.o   -c /home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:18: fatal error: ndir.h: No such file or directory
compilation terminated.
CMakeFiles/cmTC_cc8f8.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_cc8f8.dir/CheckIncludeFile.c.o' failed
make[1]: *** [CMakeFiles/cmTC_cc8f8.dir/CheckIncludeFile.c.o] Error 1
make[1]: Leaving directory '/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_cc8f8/fast' failed
make: *** [cmTC_cc8f8/fast] Error 2

Determining if the include file sys/ndir.h exists failed with the following output:
Change Dir: /home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_c655a/fast"
/usr/bin/make -f CMakeFiles/cmTC_c655a.dir/build.make CMakeFiles/cmTC_c655a.dir/build
make[1]: Entering directory '/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_c655a.dir/CheckIncludeFile.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_c655a.dir/CheckIncludeFile.c.o   -c /home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:22: fatal error: sys/ndir.h: No such file or directory
compilation terminated.
CMakeFiles/cmTC_c655a.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_c655a.dir/CheckIncludeFile.c.o' failed
make[1]: *** [CMakeFiles/cmTC_c655a.dir/CheckIncludeFile.c.o] Error 1
make[1]: Leaving directory '/home/nitish/RISCVcompiler/riscv-llvm/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_c655a/fast' failed
make: *** [cmTC_c655a/fast] Error 2