ucb-bar/esp-llvm #41: Assertion failure: "Couldn't find the register class"

luismarques, 2016-12-18:
The following LLVM IR doesn't compile to RISC-V assembly, due to an assertion failure:

```
target datalayout = "e-m:e-p:32:32:32-i1:8:16-i8:8:16-i16:16-i32:32-f32:32-f64:64-f80:128-f128:128-n32"
target triple = "riscv--linux-elf"

%bar = type { i8 }

define void @foo(%bar*, { i32, i8* }, { i32, i8* }, { i32, i8* }, i32, i32) #0 {
  %7 = alloca %bar*, align 4
  %8 = alloca { i32, i8* }, align 4
  %9 = alloca { i32, i8* }, align 4
  %10 = alloca { i32, i8* }, align 4
  %11 = alloca i32, align 4
  %12 = alloca i32, align 4
  store %bar* %0, %bar** %7
  store { i32, i8* } %1, { i32, i8* }* %8
  store { i32, i8* } %2, { i32, i8* }* %9
  store { i32, i8* } %3, { i32, i8* }* %10
  store i32 %4, i32* %11
  store i32 %5, i32* %12
  ret void
}
```

```
$ clang -target riscv-linux-elf -mriscv=RV32 -S test.ll
Assertion failed: (BestRC && "Couldn't find the register class"), function getMinimalPhysRegClass, file riscv-llvm/lib/CodeGen/TargetRegisterInfo.cpp, line 141.
Stack dump:
0.	Program arguments: /opt/riscv/bin/clang-3.9 -cc1 -triple riscv--linux-elf -S -disable-free -main-file-name test.ll -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -no-integrated-as -mconstructor-aliases -target-feature +rv32 -target-feature -rv64 -target-linker-version 274.1 -dwarf-column-info -debugger-tuning=gdb -coverage-file test.ll -resource-dir /opt/riscv/bin/../lib/clang/3.9.0 -fno-dwarf-directory-asm -fdebug-compilation-dir . -ferror-limit 19 -fmessage-length 165 -fobjc-runtime=gcc -fdiagnostics-show-option -fcolor-diagnostics -o test.s -x ir test.ll 
1.	Code generation
2.	Running pass 'Function Pass Manager' on module 'test.ll'.
3.	Running pass 'Prologue/Epilogue Insertion & Frame Finalization' on function '@foo'
```

The same error still happens when we simplify the auto generated clang/llvm call to:

```
/opt/riscv/bin/clang-3.9 -cc1 -triple riscv--linux-elf -S -main-file-name test.ll -o test.s -x ir test.ll
```

This IR is a slight simplification of the IR generated by a C-like LLVM frontend, built with this LLVM fork, branch riscv-trunk (latest commit, cd5b77c). If a fix cannot be provided in a timely fashion, workarounds would be appreciated too.

nnlight, 2017-03-03:
I had the same problem.
As I understood the issue marking functions as no-omit-framepointer should help.
Anyway attached patch helped me.
[no_reg_class_found.diff.txt](https://github.com/riscv/riscv-llvm/files/816925/no_reg_class_found.diff.txt)

luismarques, 2017-03-07:
@nnlight Thanks. I applied the patch and it didn't help me (maybe I botched it, I'm using a non-C frontend so the build steps are a bit more complicated). But the bit about the frame pointer (-disable-fp-elim) did help! Thanks!