ucb-bar/esp-llvm #43: riscv-llvm bug fixes

jeffy1009, 2017-04-03:
Hello, I want to share the bug fixes for riscv-llvm and riscv-clang.

You can find the repository with the fixes applied at the following urls:
https://github.com/jeffy1009/riscv-llvm
https://github.com/jeffy1009/riscv-clang

llvm is based on commit 871363848f6a8f02a15b4d9e167427c008012341
clang is based on commit 4cb8b5922c1abdf76bb3c0ed236ecb2fc44779d2
These commits are somewhere between LLVM 3.7 and 3.8.

The latest commits (merge with r275012/3) will break the build. 
However, it should be easy to sync with LLVM 3.8.

Some of the fixes are not that pretty. (Please refer to the commit messages)

I could also create a pull request if I have permission (I believe I don't?)

neuschaefer, 2017-04-03:
AFAIK you don't need any special permissions to create a pull request on Github. I've never been unable to do it.

jeffy1009, 2017-04-04:
Hmm.. if I try to create a pull request based on 871363848f6a8f02a15b4d9e167427c008012341, "create pull request" button disappears. I guess a pull request should be based on a branch?

If someone create a branch from that commit I could create a pull request.
I can also sync with the trunk, but I don't have time for it now..

One more thing about the bug fixes.. It's been tested with RV64. I could run all of the SPEC2000 bms. But it may still have issues with RV32.

johan-bjareholt, 2017-04-05:
@jeffy1009 That is correct, you can only pull request to a branch and not a specific commit.

However, you can still create a pull request even though it cannot automatically merge, here's an example:
https://github.com/riscv/riscv-llvm/compare/riscv-trunk...jeffy1009:jsshin

Even if you did this though the maintainer seems to be inactive so no one will be able to merge the pull request even if you do sync with riscv-trunk.

closed 2026-01-16 by jeffy1009