ucb-bar/esp-llvm #45: multiplication, i32, i64, __muldi3, __mulsi3

alexanderFell, 2017-08-10:
Hey all,
I am hoping, someone can point me out what I am doing wrong. I would like to compile the following program:

`#include <stdlib.h>
int main(int argc, char** argv) {
	short a = atoi(argv[1]);
	short b = atoi(argv[2]);
	return a * b;
}`

I am trying to compile it and it results in
`riscv64-unknown-linux-gnu-clang++ --sysroot=../../sysroot/ multiplier.cpp -S -o multiplier.S
riscv64-unknown-linux-gnu-g++ multiplier.S -o multiplier.riscv --static
/tmp/ccLOnAlk.o: In function `main':
multiplier.cpp:(.text+0x68): undefined reference to `__mulsi3'
collect2: error: ld returned 1 exit status
Makefile:26: recipe for target 'multiplier.riscv' failed
make: *** [multiplier.riscv] Error 1`

multiplier.S:
`	.text
	.file	"multiplier1.cpp"
	.globl	main
	.p2align	2
	.type	main,@function
main:
	addi	x2, x2, -64
	sd	x1, 56(x2)
	sd	x8, 48(x2)
	add	x8, x2, x0
	addiw	x5, x10, 0
	addi	x10, x11, 0
	sw	x0, 44(x8)
	sw	x5, 40(x8)
	sd	x11, 32(x8)
	ld	x11, 8(x11)
	lui	x6, %hi(atoi)
	addi	x6, x6, %lo(atoi)
	sd	x10, 16(x8)
	addi	x10, x11, 0
	sd	x6, 8(x8)
	jalr	x1, x6, 0
	sh	x10, 30(x8)
	ld	x6, 32(x8)
	ld	x10, 16(x6)
	ld	x6, 8(x8)
	jalr	x1, x6, 0
	sh	x10, 28(x8)
	lh	x5, 30(x8)
	lh	x7, 28(x8)
	addiw	x10, x5, 0
	addiw	x11, x7, 0
	jal	x1, __mulsi3
	add	x2, x8, x0
	ld	x8, 48(x2)
	ld	x1, 56(x2)
	addi	x2, x2, 64
	ret
Lfunc_end0:
	.size	main, Lfunc_end0-main`

The issue here is the __mulsi3 call, which seems to be missing in the following riscv toolchain. If I change the type of a and b to long (anything with a word size that is larger than int) __mulsi3 is replaced by __muldi3, which is defined by the RISCV toolchain and consists of a software multiplication.

I am pretty sure I made a mistake and I forgot a flag somewhere telling clang to use a hardware multiplier (mul, mulw,...), but I have no idea, what the flag should look like (I am new to llvm and riscv). I saw in RISCVInstrInfoM.td that it requires "HasM" but I do not know, where to put it.

Thanks a lot in advance,
Alex

alexanderFell, 2017-08-10:
Okay I tried the whole day, 5 minutes after submitting this issue, I figured it out:
It seems, clang didn't know, that I have a multiplier (as suspected), so changing the clang++ invocation like to:

riscv64-unknown-linux-gnu-clang++ --sysroot=../../sysroot/ multiplier.cpp -S -mriscv=RV64IAM -o multiplier.S

did the trick. Sorry for the noise created.

byllgrim, 2019-10-03:
> Sorry for the noise created.

I'm happy for this noise, as it helped me just now.