ucb-bar/esp-llvm #50: Unknown target triple error

zejiangfish, 2018-02-12:
Hi llvm maintainers,

I can build the llvm on 64-bit ubuntu, but I got error when I compile the *.c code using:
clang -target riscv64 -mriscv=RV64IAMFD -S hello.c -o hello.S

The error is:
error: unknown target triple 'riscv64', please use -triple or -arch

Did I miss something? I am on the commit 8aec2ae754fd66d4e73.

Thanks.

sorear, 2018-02-12:
If you are not a member of @ucb-bar, this is not the LLVM you should be using.  The correct LLVM for external use is the RISCV target in the upstream llvm.org repositories.

neuschaefer, 2018-02-12:
In that case this repo should perhaps be moved to https://github.com/ucb-bar/llvm, and be replaced by a staging tree for upstream LLVM

sorear, 2018-02-12:
I sent @colinschmidt an email asking for it to be moved to https://github.com/ucb-bar/hwacha-llvm since that seems to be its main function.  I don't think there is a staging repository in the sense you mean (https://github.com/lowRISC/riscv-llvm tracks the upstream port but serves a different purpose, @asb et al have been following an upstream first approach).

zejiangfish, 2018-02-12:
OK, thanks, I will try the llvm.org repo.