ucb-bar/esp-llvm #53: The target `RISCV' does not exist

TATAyu0413, 2019-12-25:
CMake Error at CMakeLists.txt:143 (message):
  The target `RISCV' does not exist.

      It should be one of

  
  AArch64;ARM;CppBackend;Hexagon;Mips;MBlaze;MSP430;NVPTX;PowerPC;Sparc;SystemZ;X86;XCore

-- Configuring incomplete, errors occurred!
See also "/home/yu/riscv-llvm/build/CMakeFiles/CMakeOutput.log".
What should I do to resolve this？Thank you so much！

pranav-prakash, 2019-12-25:
Riscv support has been merged into upstream LLVM repositories, so you should probably be using that. See #50

TATAyu0413, 2019-12-26:
> Riscv support has been merged into upstream LLVM repositories, so you should probably be using that.(See #50). If you want to play around with Hwacha vectorization support (which is still pretty janky) try the [trunk](https://github.com/ucb-bar/esp-llvm/tree/trunk) branch which builds off of a newer llvm version compared to other branches.

I get it, thanks.

closed 2019-12-26 by pranav-prakash