ucb-bar/esp-llvm #54: ld error when building esp-llvm as a release build

siyy123, 2021-01-17:
when building esp-llvm using gcc with build type release (branch: trunk), the ld would fail building target libLTO.so due to RISCVVectorFetchOptimizer refer to undefined symbol llvm::Metadata::dump() const in RISCVVectorFetchIROpt::processOpenCLKernel.
But the bug won't appear when building a debug release.

the log:

[ 76%] Linking CXX shared library ../../lib/libLTO.so
/usr/bin/ld: ../../lib/libLLVMRISCVCodeGen.a(RISCVVectorFetchOptimizer.cpp.o): in function `(anonymous namespace)::RISCVVectorFetchIROpt::processOpenCLKernel(llvm::Function*)':
RISCVVectorFetchOptimizer.cpp:(.text._ZN12_GLOBAL__N_121RISCVVectorFetchIROpt19processOpenCLKernelEPN4llvm8FunctionE+0xda9): undefined reference to `llvm::Metadata::dump() const'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/lto/CMakeFiles/LTO.dir/build.make:146: lib/libLTO.so.9svn] Error 1
make[1]: *** [CMakeFiles/Makefile2:11926: tools/lto/CMakeFiles/LTO.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

pranav-prakash, 2021-01-18:
Unless you have a specific need to target the Hwacha accelerator you should not use this. General support for riscv is present in upstream llvm.

colinschmidt, 2021-01-18:
This project isn't being actively developed so I think you will just have to settle for using the Debug build.
If you would like to fix this issue I would review a PR for it, but will not have time to fix it myself.

pranav-prakash, 2021-01-18:
You can try building the `trunk` branch instead of whatever the default is. `trunk` was rebased to a more recent version of llvm and fixes a lot of bugs in the hwacha vectorizer.

pranav-prakash, 2021-01-18:
Just to clarify did you build `riscv-trunk` (which is the default selected on github) or the (confusingly similar) `trunk` branch, which is rebased to a newer version of llvm?

siyy123, 2021-01-22:
I use the trunk branch. I think I would have a try to modify the dump codes and try again.

closed 2021-02-10 by siyy123